moveit/moveit_ros #558: moveit::planning_interface::move() API documentation

scottp11, 2015-01-23:
Hi, 

Currently the move() method blocks permanently unless there is an asynchronus spinner in use. This confuses new users (e.g. https://groups.google.com/forum/#!topic/moveit-users/IZmpzzKR2ko). Can we update the API documentation to mention this?

Current api documentation (from http://moveit.ros.org/doxygen/classmoveit_1_1planning__interface_1_1MoveGroup.html#a5dc2f2d6454084e80e3988089545a74b):
bool move() Plan and execute a trajectory that takes the group of joints declared in the constructor to the specified target. This call is always blocking (waits for the execution of the trajectory to complete). 

Possible update:
bool move() Plan and execute a trajectory that takes the group of joints declared in the constructor to the specified target. This call is always blocking (waits for the execution of the trajectory to complete) and requires an asynchronous spinner to be started. 

Regards,

Scott.

mikeferguson, 2015-01-23:
@scottp11 you can certainly open a pull request with the updates to the doxygen comments in the code -- that can get merged pretty quickly. That will be sure to update http://docs.ros.org/indigo/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html very quickly (2-3 days at most usually).

Unfortunately, the moveit.ros.org/doxygen doesn't seem to be generated anymore. @sachinchitta @isucan is there any way to kick updates on that site (last one was more than a year ago)?

sachinchitta, 2015-01-30:
The documentation from docs.ros.org is now the most up to date MoveIt! code documentation - its also accessible from http://moveit.ros.org/code-api/ - I just switched off moveit.ros.org/doxygen (which had to be hand-generated) to avoid confusion.

closed 2015-04-05 by scottpaulin