moveit/moveit_ros #559: "moveit_evaluate_collision_checking_speed" tool segfaults

skohlbr, 2015-01-24:
When I run
<pre>
rosrun moveit_ros_planning moveit_evaluate_collision_checking_speed
</pre>
I consistently get segfaults regardless of the type of robot I try. For simplicity I tried this in an environment with sim time activated (Ubuntu 14.04/AMD64/Indigo), but it's unlikely that causes the segfault I guess.

Backtrace (moveit_core compiled as Debug):
<pre>
0	__GI_raise		56	0x7f835f1dbbb9	
1	__GI_abort		89	0x7f835f1defc8	
2	__assert_fail_base		92	0x7f835f1d4a76	
3	__GI___assert_fail		101	0x7f835f1d4b22	
4	moveit::core::RobotState::getCollisionBodyTransform	robot_state.h	1032	0x7f835c02dae0	
5	collision_detection::CollisionRobotFCL::constructFCLObject	collision_robot_fcl.cpp	80	0x7f835c02c4dd	
6	collision_detection::CollisionWorldFCL::checkRobotCollisionHelper	collision_world_fcl.cpp	111	0x7f835c03ee09	
7	collision_detection::CollisionWorldFCL::checkRobotCollision	collision_world_fcl.cpp	94	0x7f835c03ecf6	
8	planning_scene::PlanningScene::checkCollision	planning_scene.cpp	522	0x7f8360c06cc1	
9	runCollisionDetection	evaluate_collision_checking_speed.cpp	51	0x431a02	
10	boost::_bi::list4<boost::_bi::value<unsigned int>, boost::_bi::value<unsigned int>, boost::_bi::value<planning_scene::PlanningScene*>, boost::_bi::value<moveit::core::RobotState*> >::operator()<void (*)(unsigned int, unsigned int, planning_scene::PlanningScene const*, moveit::core::RobotState const*), boost::_bi::list0>	bind.hpp	457	0x44312e	
11	boost::_bi::bind_t<void, void (*)(unsigned int, unsigned int, planning_scene::PlanningScene const*, moveit::core::RobotState const*), boost::_bi::list4<boost::_bi::value<unsigned int>, boost::_bi::value<unsigned int>, boost::_bi::value<planning_scene::PlanningScene*>, boost::_bi::value<moveit::core::RobotState*> > >::operator()	bind_template.hpp	20	0x44297d	
12	boost::detail::thread_data<boost::_bi::bind_t<void, void (*)(unsigned int, unsigned int, planning_scene::PlanningScene const*, moveit::core::RobotState const*), boost::_bi::list4<boost::_bi::value<unsigned int>, boost::_bi::value<unsigned int>, boost::_bi::value<planning_scene::PlanningScene*>, boost::_bi::value<moveit::core::RobotState*> > > >::run	thread.hpp	117	0x441e44	
13	??	/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0		0x7f835ff1ca4a	
14	start_thread		312	0x7f835fcfb182	
15	clone		111	0x7f835f29fefd	
</pre>

The crash happens here at BOOST_VERIFY in https://github.com/ros-planning/moveit_core/blob/indigo-devel/robot_state/include/moveit/robot_state/robot_state.h#L1032:
<pre>
  const Eigen::Affine3d& getCollisionBodyTransform(const LinkModel *link, std::size_t index) const
  {
    BOOST_VERIFY(checkCollisionTransforms());
    return global_collision_body_transforms_[link->getFirstCollisionBodyTransformIndex() + index];
  }
</pre>

Any suggestions on how to fix or avoid this are appreciated.

mikeferguson, 2015-01-24:
A real quick look shows that this might have something to do with dirty transforms (https://github.com/ros-planning/moveit_core/blob/indigo-devel/robot_state/src/robot_state.cpp#L159) -- this node is probably code that hasn't been run in a while, so I'm guessing these issues might be left over from the RobotState rewrite about a year ago. -- I've fixed numerous issues in the tests where a RobotState instance needed update() called on it (for instance: https://github.com/ros-planning/moveit_core/commit/b619e5c458e859dc6145cac1e6c8c6838ef4670d)

skohlbr, 2015-01-25:
Thanks for the hint, now works like a charm after adding update. See pull request.

closed 2015-01-25 by mikeferguson