moveit/moveit_ros #563: PointCloudOctomapUpdater consumes higher CPU than it should

skohlbr, 2015-02-04:
It seems to be that the PointCloudOctomapUpdater is using quite significant CPU although it shouldn't (especially since a largely equivalent older approach approach takes ~33% of what move_group uses, as described below).

Previously I've been using a setup with a standalone self-filter node (based on the "old" self filter") and another node creating a Octomap (which sends a region of interest to MoveIt! for planning). This combination summed up takes about 20% CPU on average.
I now tried the Octomap updater with the following settings (which are equivalent to our previous setup with the exception of a 5m range cutoff which actually should make things faster, not slower):

<pre>
sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /scan_cloud
    max_range: 5.0
    point_subsample: 1
    padding_offset: 0.025
    padding_scale: 1.0
    filtered_cloud_topic: filtered_cloud
</pre>

It takes about three times as much CPU as the previous approach, with no discernable difference in the Octomap produced. Output of "top" (uninteresting parts omitted):
<pre>
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND     
20638 kohlbre+  20   0 1440636 130924  25816 S  64,0  0,8   3:32.59 move_group  
32318 kohlbre+  20   0 1275008 183116  33880 S  14,9  1,1   0:51.13 worldmodel+  
26242 kohlbre+  20   0  732068  37788  25100 S   5,0  0,2   0:20.68 self_filter 
</pre>

Without the updater activated move_group just takes 6% CPU with the same setup started:
<pre>
30530 kohlbre+  20   0 1278444 118928  33832 S   9,6  0,7   0:28.96 worldmodel+ 
18327 kohlbre+  20   0 1416884  49208  24836 S   6,3  0,3   0:19.94 move_group  
25137 kohlbre+  20   0  732052  37756  25340 S   5,7  0,2   0:16.46 self_filter 
</pre>

"Old approach" Octomap:
[image]

PointCloudOctomapUpdater Octomap (note the is cut-off at 5m being the only difference):
[image]

Any hints on what might be the cause are appreciated.

skohlbr, 2015-02-04:
I removed all code related to the octomap update and now get about 35% CPU consumption from 'top' (essentially just from self filtering then). This is still much higher than with the old self filter, which only consumes 5% CPU.

davetcoleman, 2015-03-05:
I had a similar experience @skohlbr, updating the Octomap is very slow. I created a new plugin that does self-filtering and re-publishing of the point cloud without updating the octomap, perhaps that helps? https://github.com/ros-planning/moveit_ros/pull/530

I finished that project, however, so never bothered re-factoring it as requested in the PR. It works great, though

skohlbr, 2016-02-13:
So I just retried this using current .debs for Indigo and a custom updater that largely does the same thing as the PointCloudOctomapUpdater (https://github.com/team-vigir/vigir_manipulation_planning/tree/master/vigir_lidar_octomap_updater). Feeding two LIDAR scans at 40Hz to it I get a CPU consumption of >300% on `top` (on a Skylake i7, no less). Now trying moveit_ros from source and checking if that changes anything.

skohlbr, 2016-02-13:
Ok, so at first everything looks exactly the same when building moveit_ros from source. Then I commented out the OpenMP parallelization use here: https://github.com/ros-planning/moveit_ros/blob/indigo-devel/perception/point_containment_filter/src/shape_mask.cpp#L151

This dropped CPU consumption from 320% to ~20% ! I thus suspect that something bad happens that causes a lot of overhead when trying to parallelize that loop. In any case, I'd recommend to comment out that parallelization attempt.

mikeferguson, 2016-02-13:
Seems fixed by #644

closed 2016-02-13 by mikeferguson

skohlbr, 2016-02-17:
Btw, if somebody has an explanation for the observed high CPU load, I'd be interested to learn more ;)
I've not used OpenMP a lot, but just searched for
`#pragma omp parallel for schedule(dynamic)`
My guess is that without providing a chunk size, it defaults to 1 and the overhead of dynamic assignment of tasks to worker threads becomes extremely high. See Figure 9 here: https://software.intel.com/en-us/articles/performance-obstacles-for-threading-how-do-they-affect-openmp-code

gavanderhoorn, 2016-02-17:
@skohlbr: have you tested with some values for the chunk size? Parallelising that for-loop would be nice, and I'm assuming at some point in time it did provide a speedup.

commit 72d8edbd (2016-02-13): Merge pull request #644 from skohlbr/indigo-devel

Remove OpenMP parallelization, fixes #563
commit aff37a13 (2016-02-13): Remove OpenMP parallelization, fixes #563
commit bc2908b4 (2016-02-16): Remove OpenMP parallelization, fixes #563
commit f1609ad0 (2016-02-16): Merge pull request #645 from skohlbr/jade-devel

Remove OpenMP parallelization, fixes #563 (jade-devel)