moveit/moveit_ros #566: Adding a rviz motion planning plugin causes RViz to hang and die

TheDash, 2015-02-13:
Is there anything I need to provide? This does not work with a robot generated via MoveIt!

mikeferguson, 2015-02-13:
Are you sure all your rviz and plugin debs are up to date? If they weren't all built at the same time you might be hitting an ABI compatibility problem. I've not currently been having any problems with the motion planning plugin.

TheDash, 2015-02-13:
I did run a dist-upgrade and update beforehand.

TheDash, 2015-02-13:
Perhaps I should add shadow-fixed to my install list?

TheDash, 2015-02-17:
Sorry Mike, this was my erro. I had set Gazebo physics to "False" which caused TF not tobe published. When TF isn't published, the motion planning plugin causes RViz to hang and die.

mikeferguson, 2015-02-17:
That sounds like a possible problem -- please update the description of the ticket to be about lack of TF causing a crash (I can believe that)

TheDash, 2015-02-17:
Ok, I've updated the title of the ticket and renamed it to solved (with a tl;dr). I'm going to close it now.

closed 2015-02-17 by TheDash

davetcoleman, 2015-03-05:
Its not clear this was solved - is there not a bug with the motion planning plugin when no tf is published? This could be fixed with an error message to help future users.