moveit/moveit_ros #567: "QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called." when moving object in planning scene.

skohlbr, 2015-02-26:
When moving an mesh object in the planning scene via the ROS collision object API, eventually the following spam appears for me in the terminal:
<pre>
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
QH6232 Qhull internal error (userprintf.c): fp is 0.  Wrong qh_fprintf called.
</pre>
move_group then crashes because the stack overflows. The symptoms are also described in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=738339 and are a known problem when the file pointer in the used function is null, leading to a infinite recursion. I'm pretty sure this shouldn't happen though and that there has to be an issue on the moveit side. Note that the collision object can be moved a few times, but at some random try the described crash will happen (with a probability of ~10% per move operation or so, which makes using mesh collision objects completely unusable).
The encountered issue sounds similar to #526, but as the very visible qhull debug output is not described there, I think it might be a different one. It appears the issue only arises for mesh objects and not for primitives (which makes sense as they are already convex).

Backtrace (note I spare the reader the prior 14480 lines of infinite recursion ;) ):
<pre>
14481	qh_errexit	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8ea0ea	
14482	qh_fprintf	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8de28e	
14483	qh_errexit	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8ea0ea	
14484	qh_fprintf	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8de28e	
14485	qh_errexit	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8ea0ea	
14486	qh_fprintf	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8de28e	
14487	qh_errexit	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8e9f34	
14488	qh_fprintf	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8de28e	
14489	qh_check_points	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8c74bf	
14490	qh_new_qhull	/usr/lib/x86_64-linux-gnu/libqhull.so.6		0x2ac39e8e9627	
14491	bodies::ConvexMesh::useDimensions(shapes::Shape const*)	/opt/ros/indigo/lib/libgeometric_shapes.so		0x2ac399bf8090	
14492	bodies::Body::setDimensions(shapes::Shape const*)	/opt/ros/indigo/lib/libgeometric_shapes.so		0x2ac399bf5bea	
14493	bodies::createBodyFromShape(shapes::Shape const*)	/opt/ros/indigo/lib/libgeometric_shapes.so		0x2ac399bfd7fd	
14494	point_containment_filter::ShapeMask::addShape	shape_mask.cpp	71	0x2ac3abb8f7bf	
14495	occupancy_map_monitor::LidarOctomapUpdater::excludeShape	lidar_octomap_updater.cpp	169	0x2ac3ab8cce12	
14496	occupancy_map_monitor::OccupancyMapMonitor::excludeShape	occupancy_map_monitor.cpp	212	0x2ac398a36752	
14497	planning_scene_monitor::PlanningSceneMonitor::excludeWorldObjectFromOctree	planning_scene_monitor.cpp	749	0x2ac394c4a6a1	
14498	planning_scene_monitor::PlanningSceneMonitor::currentWorldObjectUpdateCallback	planning_scene_monitor.cpp	796	0x2ac394c4ac24	
14499	boost::_mfi::mf2<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>::operator()	mem_fn_template.hpp	280	0x2ac394ca18c2	
14500	boost::_bi::list3<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1>, boost::arg<2> >::operator()<boost::_mfi::mf2<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>, boost::_bi::list2<boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action&> >	bind.hpp	392	0x2ac394c9a723	
14501	boost::_bi::bind_t<void, boost::_mfi::mf2<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>, boost::_bi::list3<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1>, boost::arg<2> > >::operator()<boost::shared_ptr<collision_detection::World::Object>, collision_detection::World::Action>	bind_template.hpp	76	0x2ac394c915f4	
14502	boost::detail::function::void_function_obj_invoker2<boost::_bi::bind_t<void, boost::_mfi::mf2<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>, boost::_bi::list3<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1>, boost::arg<2> > >, void, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>::invoke	function_template.hpp	153	0x2ac394c865d1	
14503	boost::function2<void, boost::shared_ptr<collision_detection::World::Object> const&, collision_detection::World::Action>::operator()	function_template.hpp	767	0x2ac399333814	
14504	collision_detection::World::notify	world.cpp	234	0x2ac39933183b	
14505	collision_detection::World::addToObject	world.cpp	88	0x2ac399330da3	
14506	planning_scene::PlanningScene::processCollisionObjectMsg	planning_scene.cpp	1680	0x2ac3996c51f0	
14507	planning_scene_monitor::PlanningSceneMonitor::collisionObjectCallback	planning_scene_monitor.cpp	587	0x2ac394c496a2	
14508	boost::_mfi::mf1<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>::operator()	mem_fn_template.hpp	165	0x2ac394cabff6	
14509	boost::_bi::list2<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&> >	bind.hpp	313	0x2ac394ca444c	
14510	boost::_bi::bind_t<void, boost::_mfi::mf1<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1> > >::operator()<boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> >	bind_template.hpp	47	0x2ac394c9ccb0	
14511	boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, planning_scene_monitor::PlanningSceneMonitor, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<planning_scene_monitor::PlanningSceneMonitor*>, boost::arg<1> > >, void, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>::invoke	function_template.hpp	153	0x2ac394c95bc2	
14512	boost::function1<void, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&>::operator()	function_template.hpp	767	0x2ac394ca455b	
14513	boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const>)	function_template.hpp	153	0x2ac394c9cdb0	
14514	boost::function1<void, boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> >::operator()	function_template.hpp	767	0x2ac394ce72d8	
14515	message_filters::CallbackHelper1T<boost::shared_ptr<moveit_msgs::CollisionObject_<std::allocator<void> > const> const&, moveit_msgs::CollisionObject_<std::allocator<void> > >::call	signal1.h	76	0x2ac394ce36da	
14516	message_filters::Signal1<moveit_msgs::CollisionObject_<std::allocator<void> > >::call	signal1.h	119	0x2ac394c93df1	
14517	message_filters::SimpleFilter<moveit_msgs::CollisionObject_<std::allocator<void> > >::signalMessage	simple_filter.h	135	0x2ac394c892b5	
[... cut ...]
14536	ros::SubscriptionQueue::call()	/opt/ros/indigo/lib/libroscpp.so		0x2ac395915695	
14537	ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)	/opt/ros/indigo/lib/libroscpp.so		0x2ac3958d34f7	
14538	ros::CallbackQueue::callAvailable(ros::WallDuration)	/opt/ros/indigo/lib/libroscpp.so		0x2ac3958d4303	
14539	ros::AsyncSpinnerImpl::threadFunc()	/opt/ros/indigo/lib/libroscpp.so		0x2ac395918024	
14540	??	/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0		0x2ac39a847a4a	
14541	start_thread		312	0x2ac3963d8182	
14542	clone		111	0x2ac396e0d00d	
</pre>

skohlbr, 2015-02-26:
Ok, after comparing backtraces it looks like this is a duplicate of #526 after all, with the difference being that for me the infinite recursion in qhull happens.

dornhege, 2015-12-11:
This seems reasonable. When #526 happens, the result is that the file pointer gets 0. Usually this leads to a SIGSEGV, but it might have been caught somewhere in qhull, although not properly, causing this error message and eventually stackoverflow.

Does the pull request fix this for you, so that this issue can be closed?

skohlbr, 2015-12-12:
Yes, this works with the PR. Thanks for looking into it!

closed 2015-12-14 by dornhege