moveit/moveit_ros #572: benchmarks missing tf dependency

mryellow, 2015-03-30:
It appears `moveit_ros_benchmarks` is missing a dependency for `tf`.

My environment is a little out of the ordinary which might be why this hasn't come up earlier, debian jessie with mostly source installs in a parent catkin workspace, jessie has few pre-compiled packages.

Was getting not found errors on `tf/tf.h` compiling `moveit_ros_benchmarks`, patching it's `package.xml` and `CMakeLists.txt` to include `tf` worked.

davetcoleman, 2015-03-30:
Perhaps some change elsewhere revealed this discrepancy. It can't hurt to add it either way - could you put in a PR?

mryellow, 2015-03-30:
Will do, gotta get some "work work" done, will hit it this-afternoon sometime.

commit 65f9e93e (2015-03-31): Adding tf dep fixes #572
commit 9c0fb659 (2015-03-31): Merge pull request #573 from mryellow/fix-572

Adding tf dep fixes #572
closed 2015-03-31 by davetcoleman