moveit/moveit_ros #577: Configurable tolerance values for pick/place

corot, 2015-04-18:
Hi, I'm experiencing problems described [here](https://groups.google.com/forum/#!topic/moveit-users/-Eie-wLDbu0) and updated [here](https://groups.google.com/forum/#!topic/moveit-users/Jh04El6-5Wc) with a hackish solution.
In sort, a limited-dof robot as the turtlebot arm cannot choose the 6D pose where to pick and place, but the tolerance values are fixed hardcoded default values in [this function](https://github.com/ros-planning/moveit_core/blob/indigo-devel/kinematic_constraints/include/moveit/kinematic_constraints/utils.h#L115). 
So place always fails unless you calculate very carefully the the yaw angle of the target pose. Should not move_group provide a way to configure tolerance values?

corot, 2015-11-15:
I still think the configurable values would be a good thing, but now I think a better solution would be to make [this line](https://github.com/ros-planning/moveit_ros/blob/jade-devel/manipulation/pick_place/src/place.cpp#L64) optional. That is, allow place goals just for the end-effector, ignoring the final pose of the attached object. I think this makes a lot of sense for placing objets whose final orientation is irrelevant, as spheres. And in my cases, for limited-dof arms that cannot choose  the final pose in most situations.

So I rename the issue!

davetcoleman, 2015-11-17:
+1, I remember this was very misleading and confused me when I was using the pick and place pipeline... currently you have to specify the start state in the end effector from of reference, but the goal state in the object frame of reference. i guess we'd need to have an option that switches between the two modes