moveit/moveit_ros #579: Add ability to monitor asynchronous execution

adzenith, 2015-05-07:
I'd like to be able to asynchronously execute a plan, and then monitor whether the plan is still running (or get a callback when it's done; either way).

The only info I've seen on this topic is from last year, saying it's impossible:
https://groups.google.com/forum/#!topic/moveit-users/LQGu9rKY10E

Is that still true? I was looking through the code, and I don't see anything that's exposed within the move group.

Thanks!

locke87, 2016-02-09:
I currently stumbled upon the same issue. Is there a chance that a Pull-Request on this topic would be accepted? My attempt would be to create an ExecuteActionCapability and use that instead of the ServiceCapability inside the move_group. Any objections against this approach?

I guess the timeline would be at least 2-3 months, I guess.

david-brodeur, 2016-07-22:
What I did is to subscribe to /joint_trajectory_action/result. 

From my understanding, moveit uses actions which are similar to ROS services in the sense that when you publish a goal, you also receive a result. So even though asyncExecute() ends before the execution of the trajectory is over, you will still receive a /joint_trajectory_action/result when it will be over. Just subscribe to it and raise a flag in your callback. 

That is best solution I have found so far and it lets me read execute other stuff while executing the trajectory. When it is done, I take the newest command received and plan/execute a new trajectory. 

I am still testing this solution while integrating/optimizing my ROS node that wraps moveit. I'll try to keep you in touch if I find anything better.

Regards,

David

v4hn, 2016-07-22:
On Fri, Jul 22, 2016 at 06:29:07AM -0700, david-brodeur wrote:
> What I did is to subscribe to /joint_trajectory_action/result. 
> 
> That is best solution I have found so far

This is obviously an ugly work-around.
If MoveIt would use a different FollowJointTrajectory action server, you would not receive a result anymore and it might take quite a while to debug this then.

We are currently working on an Action to execute a known RobotTrajectory.
In the long run this will hopefully replace the current ExecuteService.
See https://github.com/ros-planning/moveit_ros/pull/719