moveit/moveit_ros #583: Is move_group communication racy?

ndantam, 2015-06-02:
The example code using move_group contains sleeps after the scene is updated (https://github.com/ros-planning/moveit_pr2/blob/hydro-devel/pr2_moveit_tutorials/planning/src/planning_scene_ros_api_tutorial.cpp).  Are these sleeps necessary for correct operation?  It seems so and that one cannot guarantee ordering on messages sent to move_group.  Without the sleeps, a planning request could be processed before the scene is updated, effectively ignoring the updates to the scene.

A total ordering is needed on all messages to move_group.  This could be done by switching from pub/sub to a remote procedure call style of communication or by pushing all pub/sub messages to move_group though the same topic using an ordered transport (e.g., TCP).

skohlbr, 2016-02-13:
Yes, this appears to be the case. I guess providing a ROS Service API for most operations such as modifying the planning scene would allow for fixing some of the issues encountered (as clients could use those services, which are blocking until they have been executed). Waiting for results could also be done using Actions (but little more involved in terms of implementation).

v4hn, 2016-06-13:
See https://groups.google.com/forum/#!topic/moveit-users/D7MWZEUSKLc for further discussion
and https://github.com/ros-planning/moveit_ros/pull/686 for a solution to this problem.
I suppose this can be closed once #686 is merged

closed 2016-07-04 by v4hn