moveit/moveit_ros #588: Compile error: ‘struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’

simonschmeisser, 2015-06-29:
Hi

I'm running indigo-devel on indigo and just pulled the latest commits. Now I get compile errors, I guess some changes should not have been backported from jade-devel?

```
[ 35%] Building CXX object moveit_ros/robot_interaction/CMakeFiles/moveit_robot_interaction.dir/src/kinematic_options.cpp.o
In file included from /usr/include/boost/math/policies/policy.hpp:20:0,
                 from /usr/include/boost/math/policies/error_handling.hpp:19,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/indigo/include/ros/time.h:58,
                 from /opt/ros/indigo/include/ros/serialization.h:34,
                 from /opt/ros/indigo/include/geometry_msgs/PoseStamped.h:14,
                 from /opt/ros/indigo/include/moveit/kinematics_base/kinematics_base.h:40,
                 from /home/ros/ros_upstream/src/moveit_ros/robot_interaction/include/moveit/robot_interaction/kinematic_options.h:40,
                 from /home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:37:
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp: In member function ‘void robot_interaction::KinematicOptions::setOptions(const robot_interaction::KinematicOptions&, robot_interaction::KinematicOptions::OptionBitmask)’:
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:119:3: error: invalid application of ‘sizeof’ to incomplete type ‘boost::STATIC_ASSERTION_FAILURE<false>’
   BOOST_STATIC_ASSERT(sizeof(kinematics::KinematicsQueryOptions) ==
   ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:119:3: error: template argument 1 is invalid
   BOOST_STATIC_ASSERT(sizeof(kinematics::KinematicsQueryOptions) ==
   ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:120:59: error: invalid type in declaration before ‘;’ token
                       sizeof(DummyKinematicsQueryOptions));
                                                           ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:94:12: error: ‘struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
     F(int, discretization_method, DISCRETIZATION_METHOD)
            ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:136:22: note: in definition of macro ‘F’
             options_.member = source.options_.member;
                      ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:137:3: note: in expansion of macro ‘QO_FIELDS’
   QO_FIELDS(F)
   ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:94:12: error: ‘const struct kinematics::KinematicsQueryOptions’ has no member named ‘discretization_method’
     F(int, discretization_method, DISCRETIZATION_METHOD)
            ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:136:47: note: in definition of macro ‘F’
             options_.member = source.options_.member;
                                               ^
/home/ros/ros_upstream/src/moveit_ros/robot_interaction/src/kinematic_options.cpp:137:3: note: in expansion of macro ‘QO_FIELDS’
   QO_FIELDS(F)
   ^
```

simonschmeisser, 2015-07-01:
this seems to be due to commits in 58c4aa93290207245d1f718b91248c76ff75df4a which I assume should not have landed in branch indigo-devel

jonbinney, 2015-07-02:
Yeah, doesn't compile for me either. I had to roll back to before that commit to get it to compile.

simonschmeisser, 2015-07-08:
fixed by 937a6f0bbaa90b32c544709e709de14f542a8895

thanks

closed 2015-07-08 by simonschmeisser