moveit/moveit_ros #591: Adding service wrapper to give easy access to warehouse function

dg-shadow, 2015-07-13:
We (Shadow Robot) need to access to warehouse from moveit_commander for one of our projects. I plan to do this by adding service interfaces to various warehouse functions. Would this be of interest as a contribution for this package? I am starting work on this more or less now, so if anyone has any comments/suggestions/flame, they'd be appreciated but get them in soon please :)

shaun-edwards, 2015-07-15:
+0....I'm all for contributing and capability to MoveIt.  However, I don't know enough to say whether what you propose is useful or appropriate.  Hopefully someone more knowledgeable will chime in!

davetcoleman, 2015-07-15:
Could you explain more why you would want this? You'd like to load saved collision objects and planning scenes from Warehouse using the Commander scripting language?

dg-shadow, 2015-07-17:
Basically yes, although it's actually robot states I need first. I wish to be able to load, manipulate and use the states from within a python node. I figured services were a nice and general way to get access to the Warehouse contents.

davetcoleman, 2015-07-31:
Sounds like a good idea. I've never much liked the mongodb (that's what it is, right?) backend of warehouse, but in theory these are very useful features, especially from Commander.