moveit/moveit_ros #592: MoveGroup class_loader::LibraryUnloadException on destruction.

Kukanani, 2015-07-13:
I am getting the following error when destroying a MoveGroup object:

    terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
      what():  Attempt to unload library that class_loader is unaware of.
    Aborted (core dumped)

With the following trivial example:

```c++
#include <moveit/move_group_interface/move_group.h>
int main(int argc, char** argv)
{
  ros::init(argc, argv, "test_motion");
  moveit::planning_interface::MoveGroup moveGroup("manipulator");
}
```

oliverek12, 2015-11-15:
I am not sure of the underlying reason for this issue, but placing a ros::spin() at the end of the main loop will keep the movegroup from throwing the exception which crashes the node.

Kukanani, 2015-11-16:
This fix doesn't make much sense for me, since I am getting the error when my node shuts down. Why add a spin() at the end of my code?

fontysrobotics, 2016-04-05:
I'm also running in the same problem, but I'm following the example code of the ROS Industrial tutorial http://aeswiki.datasys.swri.edu/rositraining/indigo/Exercises/3.6

There the node looks like this one:

```
#include <ros/ros.h>
#include <moveit/move_group_interface/move_group.h>
int main(int argc, char **argv)
{
  ros::init(argc, argv, "lesson_move_group");

  // start a background "spinner", so our node can process ROS messages
  //  - this lets us know when the move is completed
  ros::AsyncSpinner spinner(1);
  spinner.start();

  moveit::planning_interface::MoveGroup group("manipulator");
  group.setRandomTarget();
  group.move();
}
```

dbolkensteyn, 2016-06-02:
This issue is linked from here: http://answers.ros.org/question/218945/a-trouble-with-the-training-unit-36-from-ros-i/

PR #706 (2016-07-07): don't destroy SharedStorage
…  to avoid interference with destruction of class_loader's static variables
This is actually a workaround for design issues in [class_loader](https://github.com/ros/class_loader/pull/34#issuecomment-205002900).
Fixes #592. Replaces #702 and #705.
closed 2016-07-07 by v4hn