moveit/moveit_ros #623: MoveIt with source installed OMPL

amdshameer, 2015-11-09:
I have a source installation of OMPL. How do I set MoveIt! to use the source installation of ompl? It works only when I do "sudo apt-get install ros-hydro-ompl". It does not find OMPL, If I remove "ros-hydro-ompl".

amdshameer, 2015-11-09:
I solved it. You don't need catkin_make_isolated. I followed http://ompl.kavrakilab.org/buildSystem.html Just follow the steps below:

1st step:
Install MoveIt from source
After successful installation of MoveIt from source, do "sudo apt-get remove ros-hydro-ompl" and "sudo apt-get purge ros-hydro-ompl"
path: /home/name/moveit

2nd step:
git clone https://github.com/ompl/ompl.git
path: /home/name/ompl
cd ompl/
mkdir -p build/Release
cd build/
cmake -DCMAKE_INSTALL_PREFIX=/usr/local ..
make install
sudo make install
cd Release/
cmake -DCMAKE_INSTALL_PREFIX=/usr/local ..
cd ..
make -j 4 update_bindings
make 
sudo make install

So this made me use source built OMPL with my youBot moveit package(catkin package).
Also DON'T FORGET TO export CMAKE_PREFIX_PATH="/home/name/project_ws/devel:/opt/ros/hydro:/home/name/moveit/devel"

@isucan @davetcoleman @sachinchitta

closed 2015-11-09 by amdshameer

liuyouzhao, 2017-04-24:
What you tried are very helpful to me, thanks very much.

davetcoleman, 2017-04-25:
See also: http://moveit.ros.org/install/source/dependencies/