moveit/moveit_ros #631: Indigo and Jade moveit_ros Release

rethink-imcmahon, 2015-12-17:
I'm concerned that we haven't seen a release of MoveIt since January 24th of this year. With nearly a years worth of updates and bugfixes, could we release MoveIt up to Bloom for the next sync? I'd even volunteer to do it, given the permissions / MoveIt specific Bloom instructions... Major bugfixes include:

TrajectoryVisualization Segfault Fix (Sept): https://github.com/ros-planning/moveit_ros/pull/611
Service Interface Access for Robot States in the db (July): https://github.com/ros-planning/moveit_ros/pull/594
Mimic Joint Support (April): https://github.com/ros-planning/moveit_ros/pull/575
Collision Check Crash Fix (Jan): https://github.com/ros-planning/moveit_ros/pull/560

And [__many__ others...](https://github.com/ros-planning/moveit_ros/pulls?page=1&q=merged%3A%3E%3D2015-01-24&utf8=%E2%9C%93)

mikeferguson, 2015-12-17:
Please see https://github.com/ros-planning/moveit_core/issues/270

rethink-imcmahon, 2015-12-17:
Aha! Right you are, Mike. It appears that I am late to the game here... Alrighty, I'll move the discussion over there. A big +1 in support of a release soon.

mikeferguson, 2015-12-17:
The jade release is also a bit more complex -- because Ubuntu broke us quite badly. There are a series of pull requests that still need testing before we can merge those and then continue releasing. I actually get two weeks off starting next week and plan to do a major pass through all of those and tag new releases -- although I may wait a few days before blooming the new releases to make sure we have a full 3-4 weeks of soak time before the debs get synced so that people can test.

rethink-imcmahon, 2015-12-17:
Sure, that makes sense. I would be in favor of release Indigo separately (sooner) then, as it would give most of the shared changes some testing time in advance of the Jade specific (potentially breaking) changes.

mikeferguson, 2015-12-17:
The one thing I really want to make sure we have is soak time since we have not released in almost a year. That does mean we should wait for the next sync to happen and then bloom the day after (we can tag releases ahead of that, but the actual update to the rosdistro should really be done just a day or two after a sync for the max amount of shadow-fixed soak time). Along the same line of thinking -- if we actually have our soak over the holiday weeks, we might not actually get any testing.

rethink-imcmahon, 2015-12-17:
I see where you are coming from, but I think the last year of people building from source has been a large soak test in itself. The bug fixes here are so numerous that I find using the Debian install too debilitating to use at all. Probably the most egregious of the issues, a robot with any mimic joints will fail to load entirely until the mimic joint's state is published. Can we put a call out to the community to help us test so that we can cut a release?

davetcoleman, 2015-12-17:
I think we are being too conservative with our releases - as @rethink-imcmahon hints, most people who actually use MoveIt! are building from source because the debs are so behind (and because so many features are missing)

130s, 2016-01-19:
Now that [the sync is done](http://lists.ros.org/pipermail/ros-users/2016-January/069795.html), can we get the ball rolling?

rethink-imcmahon, 2016-01-28:
The new [docker-based buildfarm was deployed ](http://lists.ros.org/pipermail/ros-users/2016-January/069824.html) last week. A MoveIt release at this point would still have a decent soak-in period before the next sync. Is there anything we can do to help the release along now?

sachinchitta, 2016-01-29:
If there's no objections on the mailing list, @130s, @rethink-imcmahon, please go ahead and do a release and coordinate with anyone else available. Thank you for your help!

mikeferguson, 2016-02-13:
Updating description to just say "jade" as indigo has been updated but jade remains a small mess

mikeferguson, 2016-06-09:
I'm currently releasing 0.6.6 into jade to address the regressions (and build farm blockage) noted in https://github.com/ros-planning/moveit_ros/issues/681 -- something I just noticed is that indigo is apparently at 0.7.1 -- so we probably want to push jade/kinetic to 0.8 on the next release (would be good to coincide this with the mongo fixes which will cause a somewhat large API/ABI break)

130s, 2016-06-09:
Looking at changelogs, probably it was `moveit_core` that made us bump the minor version to 0.7.x and `moveit_ros` followed that
http://docs.ros.org/indigo/changelogs/moveit_core/changelog.html
http://docs.ros.org/indigo/changelogs/moveit_ros/changelog.html

For Jade, [this change](https://github.com/ros-planning/moveit_ros/commit/1ecc88bf77556a456385e39078ab6ae055a8ae30) differentiates from indigo so I thought we better bump minor version to 0.8.x. In fact `moveit_core` is already 0.8.x
http://docs.ros.org/jade/changelogs/moveit_core/changelog.html

mikeferguson, 2016-06-09:
I think the whole bump to 0.7 was probably just done wrong -- indigo had been 0.6 for the first 18 months of releases, and then suddenly jumped to 0.7, while jade should probably have been 0.7 but has continued to be released from the 0.6 branch.

Regardless, I'm going to leave moveit_ros at 0.6.6 for the time being -- since it took something like 30 minutes to bloom. We still won't be fully building (warehouse ros will cause problems on >trusty), so as we address that we can go to 0.8.x here.

130s, 2016-06-09:
> I think the whole bump to 0.7 was probably just done wrong -- indigo had been 0.6 for the first 18 months of releases, and then suddenly jumped to 0.7,

I see.
Looks like Hydro stayed withing 0.5, so Indigo could have kept going in 0.6.
http://docs.ros.org/hydro/changelogs/moveit_core/changelog.html
http://docs.ros.org/hydro/changelogs/moveit_ros/changelog.html

> I'm going to leave moveit_ros at 0.6.6 for the time being -- since it took something like 30 minutes to bloom. We still won't be fully building (warehouse ros will cause problems on >trusty), so as we address that we can go to 0.8.x here.

+1

davetcoleman, 2016-08-04:
@130s can this be closed/consolidated?

mikeferguson, 2016-08-24:
This should be good enough to close now

closed 2016-08-24 by mikeferguson