moveit/moveit_ros #632: Any way to control acceleration of generated trajectories for individual requests?

hemes, 2016-01-06:
I have recently started using moveit and am working though how to control speed and acceleration of the generated trajectories.  I came across PR #547 that describes the max_velocity_scaling_factor and was wondering if it would make sense to have a similar scaling factor for acceleration.  I am using this parameter (and it is great!) but I still have some cases where I am indifferent to the speeds but need less acceleration to get smooth motion (e.g., heavy and/or fragile payloads).  Plus I think having both speed and acceleration scaling makes sense from a consistency standpoint.

Currently I am just making changes to the joint_limits.yaml file but the ability to control this on a per trajectory basis would be awesome!

davetcoleman, 2016-01-06:
I had this same thought recently - it would be nice to have an acceleration scalar as well. I think you could mimic the changes in this PR https://github.com/ros-planning/moveit_ros/pull/547 for acceleration and submit the change as a new PR

hemes, 2016-01-07:
Agreed.  I will see if I can find time to do this over the next few weeks.

hemes, 2016-01-12:
Submitted via PR #634

davetcoleman, 2016-01-13:
Sweet!

davetcoleman, 2016-02-04:
All 3 PRs have been merged. Re-open this issue if needed.

closed 2016-02-04 by davetcoleman