moveit/moveit_ros #638: Duplicate moveit_ros.py

130s, 2016-01-19:
Maybe a minor issue though.

Asked in https://github.com/ros-planning/moveit_ros/pull/584#issuecomment-173017776 originally.
`moveit_joy.py` gets installed in 2 different locations after #584.

```
-rw-r--r-- 1 n130s n130s 23K Jun 19  2015 ./install/lib/moveit_ros_visualization/moveit_joy.py
-rw-rw-r-- 1 n130s n130s 23K Jun 19  2015 ./install/lib/python2.7/dist-packages/moveit_ros_visualization/moveit_joy.py
```

I may have found why #584 is needed; `roslaunch YOURROBOT_moveit_config joystick_control.launch` (in [visualization/doc/joystick.rst](https://github.com/ros-planning/moveit_ros/blob/784f6aeabc282c2e5a80bb01884c7e6d6830d126/visualization/doc/joystick.rst)) fails without #584.

But if this is correct, distutils-based installation by `setup.py` is not working. We don't want to clutter install space with duplicate files.

130s, 2016-01-20:
I think ideally what we want is a python module that can also be called via `rosrun`. I [asked how to do so](http://answers.ros.org/question/224546/call-python-module-as-a-script-in-catkin/) by Catkin (maybe keep using distutils).

130s, 2016-07-03:
A couple of PRs to address this issue exist.

- #703 (Everything seems to conform to ROS buildfarm maintainers' suggestion). The launch files that have been already generated might not work with #703.
- #704 maintains backward compatibility by keeping the script file name as `moveit_joy.py` ([file extension not recommended for scripts](http://docs.ros.org/jade/api/catkin/html/howto/format2/installing_python.html)).

I'm file for either way but leaning toward guaranteed backward compatibility #704.

davetcoleman, 2016-07-05:
The ROS guidelines say "executable script names **generally** do not include a .py suffix" but I think we have a good exception to this general rule. If we can name them both moveit_joy.py as @v4hn recommended that would be best

v4hn, 2016-07-21:
resolved in #704

closed 2016-07-21 by v4hn