moveit/moveit_ros #649: URL not working

betsch85, 2016-02-22:
http://docs.ros.org/indigo/api/pr2_moveit_tutorials/html/planning/src/doc/move_group_interface_tutorial.html#compiling-the-code

The link under "Compiling the Code" links to http://moveit.ros.org/wiki/Installation, which is not working (same for other C++ tutorials listed under http://moveit.ros.org/documentation/tutorials/)

alainsanguinetti, 2016-07-05:
In my case the links for "Compiling the code" from the two urls you mentioned redirect to [http://moveit.ros.org/install/](http://moveit.ros.org/install/).
Maybe you should refresh your browser cache ? Or try opening those pages from a private tab / window to see if it is a problem with your browser cache.

v4hn, 2016-07-05:
This was fixed in https://github.com/ros-planning/moveit_pr2/pull/69 ,
thanks for pointing out this request @alainsanguinetti!
Closing as resolved

closed 2016-07-05 by v4hn