moveit/moveit_ros #653: Parse error at "BOOST_JOIN" with Boost 1.58 in moveit_ros/visalization package

enricovillagrossi, 2016-03-11:
ERROR:
usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN" 

appear compiling visulization package.  
It seems to be a known problem as reported here: 
https://github.com/stonier/qt_ros/issues/8
https://github.com/ros-planning/moveit_ros/pull/399

already fixed after 0.5.13 (2014-02-06), but I still have the same problem, some workarounds proposed:
https://bugreports.qt.io/browse/QTBUG-22829  

don't work for me. I'm trying to compiling both indigo-devel and jade-devel under debian/stretch with Boost 1.58 and Qt4 4.8.
Can anyone help me?
Thank you in advance!

Enrico

nikhilkalige, 2016-04-05:
@enricovillagrossi 

Make the following changes in files
`trajectory_rviz_plugin/include/..../trajectory_display.h`

```
#ifndef MOVEIT_TRAJECTORY_RVIZ_PLUGIN__TRAJECTORY_DISPLAY
#define MOVEIT_TRAJECTORY_RVIZ_PLUGIN__TRAJECTORY_DISPLAY

#include <rviz/display.h>

#include <moveit/rviz_plugin_render_tools/trajectory_visualization.h>
#ifndef Q_MOC_RUN
#include <ros/ros.h>
#include <moveit/rdf_loader/rdf_loader.h>
#endif
```

`rviz_plugin_render_tools/include/..../trajectory_visualization`
```

#ifndef MOVEIT_TRAJECTORY_RVIZ_PLUGIN__TRAJECTORY_VISUALIZATION
#define MOVEIT_TRAJECTORY_RVIZ_PLUGIN__TRAJECTORY_VISUALIZATION

#include <moveit/macros/class_forward.h>
#include <rviz/display.h>

#ifndef Q_MOC_RUN
#include <moveit/rviz_plugin_render_tools/robot_state_visualization.h>
#include <ros/ros.h>
#include <moveit/robot_model/robot_model.h>
#include <moveit/robot_state/robot_state.h>
#include <moveit/robot_trajectory/robot_trajectory.h>
#include <moveit_msgs/DisplayTrajectory.h>
#endif

namespace rviz
```
I was able to compile after this, but I am not sure if that is the right way of adding `#ifndef`.

enricovillagrossi, 2016-04-06:
Thank you @nikhilkalige ! 
Your suggestion fix the problem.

Enrico

davetcoleman, 2016-04-12:
pull request?

commit 07701de7 (2016-04-28): Merge pull request #666 from Thomqa/jade-devel

Solved parse error with Boost 1.58. Fixes #653
commit ccdb4b07 (2016-04-28): Solved parse error with Boost 1.58. Fixes #653
closed 2016-04-28 by davetcoleman