moveit/moveit_ros #654: fails to compile with the latest version of rviz

wjwwood, 2016-03-23:
For example:

http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__moveit_ros_benchmarks_gui__ubuntu_trusty_amd64__binary/10/console

I believe the root cause of this is that the "default_plugin" is no longer included in rviz's `_LIBRARIES` variable, but `moveit_ros_benchmark_gui` uses headers from the default plugin and so it needs to link against them. There is now a cmake variable called `rviz_DEFAULT_PLUGIN_LIBRARIES` which you can add to your `target_link_libraries` which will contain the default plugin. I recommend that this variable be used to fix the current compilation problems.

See:

- https://github.com/ros-visualization/rviz/pull/948
- https://github.com/ros-visualization/rviz/issues/964
- https://github.com/ros-visualization/rviz/pull/979

dornhege, 2016-03-30:
Partially solved with #657 and #658. I'm not sure if this can be solved for catkin_make without any changes in rviz.

wjwwood, 2016-03-30:
Can anyone try with https://github.com/ros-visualization/rviz/pull/983?

dornhege, 2016-03-31:
This doesn't fix it for me. The problem is that the rviz extras wants the generated file and thus fails before linking during cmake already.

```
CMake Error at .../devel/share/rviz/cmake/rviz-extras.cmake:2 (include):
  include could not find load file:
.../devel/share/rviz/cmake/default_plugin_location.cmake
Call Stack (most recent call first):
  .../devel/share/rviz/cmake/rvizConfig.cmake:190 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
moveit_ros/benchmarks_gui/CMakeLists.txt:19 (find_package)
```

wjwwood, 2016-03-31:
I see, let me try to address that.

wjwwood, 2016-03-31:
I added https://github.com/ros-visualization/rviz/pull/983/commits/91ba11a4bc18c55a857a10e61e975e8cebc8c052 to try and address that. Can you give that a try?

dornhege, 2016-04-01:
It did compile with catkin_make. However only after multiple runs with `--force-cmake`. On the first run I'm getting linker errors, because it's not linking the library.

wjwwood, 2016-04-01:
@dornhege is that from a clean build it fails on the first pass?

dornhege, 2016-04-01:
Yes, after I removed build/ and devel/.

wjwwood, 2016-04-01:
@dornhege can you send me a `.rosinstall` file for your workspace?

dornhege, 2016-04-01:
The .rosinstall is attached, where I manually checked out your rviz commit.
After a second pass with `catkin_make --force-cmake` it build.

[.rosinstall.txt](https://github.com/ros-planning/moveit_ros/files/200434/default.rosinstall.txt)

wjwwood, 2016-04-01:
I can reproduce that issue and I think I understand why it is a problem. I changed the way I expose it with https://github.com/ros-visualization/rviz/pull/983/commits/610ecf5c9df5b5e9a3263a68de8ac4d4eca59446. I'll just have to remember to update that when I change the default plugin's library name in Kinetic.

I'm rebuilding now to test it.

wjwwood, 2016-04-01:
It seems to build for me now. I'm gonna merge the rviz pr and prepare for another release. Let me know if you see anymore issues with it. Thanks for helping me test it.

closed 2016-04-01 by wjwwood

dornhege, 2016-04-01:
Nice work. Builds fine for me to.

wjwwood, 2016-04-02:
@dornhege FYI, I have made a slight improvement after thinking about it more and seeing a failure during one of the repeated builds of the moveit workspace: https://github.com/ros-visualization/rviz/pull/984

130s, 2016-04-07:
Is this resoloved? I'm still seeing [a build failure](http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__moveit_ros_benchmarks_gui__ubuntu_trusty_amd64__binary/25/console).

wjwwood, 2016-04-07:
@130s I'm not sure. I can only verify that I can build @dornhege's workspace with both `catkin_make` and `catkin_make_isolated`.

k-okada, 2016-04-08:
> I'm still seeing a build failure.

because 0.7.0 does not include https://github.com/ros-planning/moveit_ros/commit/35ce06cc49c3036d436fa32ed24cc684f5bff462