moveit/moveit_ros #656: Fails to compile under Jade due to changed controller_manager_msgs structure

barryridge, 2016-03-29:
Hi all,

I have been having trouble compiling MoveIt on Ubuntu 14.04 under ROS Jade.  The compilation bugs out with the following error:

> Linking CXX shared library /home/reconcell/moveit/deve/lib/libmoveit_controller_manager_example.so
> [ 65%] Built target moveit_controller_manager_example
> Scanning dependencies of target moveit_ros_control_interface_trajectory_plugin
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘void moveit_ros_control_interface::MoveItControllerManager::allocate(const string&, const ControllerState&)’:
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:153:52: error: ‘const ControllerState’ has no member named ‘resources’
>            alloc_it->second->alloc(name, controller.resources);  // allocate handle
>                                                     ^
> [ 67%] /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘virtual void moveit_ros_control_interface::MoveItControllerManager::getControllerJoints(const string&, std::vector<std::basic_string<char> >&)’:
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:244:27: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
>        joints = it->second.resources;
>                            ^
> Building CXX object moveit_plugins/moveit_ros_control_interface/CMakeFiles/moveit_ros_control_interface_trajectory_plugin.dir/src/joint_trajectory_controller_plugin.cpp.o
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘virtual bool moveit_ros_control_interface::MoveItControllerManager::switchControllers(const std::vector<std::basic_string<char> >&, const std::vector<std::basic_string<char> >&)’:
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:286:61: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
>        for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
>                                                              ^
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:286:95: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
>        for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
>                                                                                                ^
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:311:63: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
>          for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
>                                                                ^
> /home/reconcell/moveit/src/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:311:97: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
>          for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
>                                                                                                  ^
> 

I did a little more investigating and it would seem that the structure of controller_manager_msgs has changed between Indigo and Jade, such that ControllerState.msg has gone from this:
`string name`
`string state`
`string type`
`string hardware_interface`
`string[] resources`

to this:

`string name`
`string state`
`string type`
`controller_manager_msgs/HardwareInterfaceResources[] claimed_resources`

where HardwareInterfaceResources.msg now contains the resources field:

`# Type of hardware interface`
`string hardware_interface`
`# List of resources belonging to the hardware interface`
`string[] resources`

I will revert back to Indigo for now and will try building MoveIt from there, but I hope the above might be helpful for developing the Jade release.  I would try to fix it myself, but I'm not familiar enough with the codebase and don't have the time right now.

Cheers,
Barry

davetcoleman, 2016-03-30:
Please try my fix (that's been an open PR for 2 months):
https://github.com/ros-planning/moveit_plugins/pull/15

barryridge, 2016-03-31:
@davetcoleman Thanks for the response.  I may not get a chance to test this, since we've reverted to Indigo on the boxes we're using here, but if I do, I'll post back here.

Thomqa, 2016-04-29:
This should be fixed right now. It does compile on Jade. Thanks @davetcoleman !

davetcoleman, 2016-04-29:
I merged it yesterday.

closed 2016-04-29 by davetcoleman