moveit/moveit_ros #662: indigo release

dornhege, 2016-04-08:
Should we release 0.7.1 for indigo to fix the current build after rviz changes? See #654 and #660.

@mikeferguson

archielee, 2016-04-12:
+1 need for PR2

130s, 2016-04-12:
Prerelease test fails like this. Does anyone have an idea?

```
mkdir -p /tmp/prerelease_job
cd /tmp/prerelease_job
generate_prerelease_script.py \
  https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml \
  indigo default ubuntu trusty amd64 \
  moveit_ros \
  --level 1 \
  --output-dir ./

$ ./prerelease.sh 
Prerelease script
:
[ 54%] Building CXX object planning_request_adapter_plugins/CMakeFiles/moveit_default_planning_request_adapter_plugins.dir/src/add_time_parameterization.cpp.o
/tmp/catkin_workspace/src/moveit_ros/planning/planning_request_adapter_plugins/src/add_time_parameterization.cpp: In member function ‘virtual bool default_planner_request_adapters::AddTimeParameterization::adaptAndPlan(const PlannerFn&, const PlanningSceneConstPtr&, const MotionPlanRequest&, planning_interface::MotionPlanResponse&, std::vector<long unsigned int>&) const’:
/tmp/catkin_workspace/src/moveit_ros/planning/planning_request_adapter_plugins/src/add_time_parameterization.cpp:160:97: error: ‘const MotionPlanRequest’ has no member named ‘max_acceleration_scaling_factor’
       if (!time_param_.computeTimeStamps(*res.trajectory_, req.max_velocity_scaling_factor, req.max_acceleration_scaling_factor))
                                                                                                 ^
make[2]: *** [planning_request_adapter_plugins/CMakeFiles/moveit_default_planning_request_adapter_plugins.dir/src/add_time_parameterization.cpp.o] Error 1
make[1]: *** [planning_request_adapter_plugins/CMakeFiles/moveit_default_planning_request_adapter_plugins.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'moveit_ros_planning': 
  Command '['/tmp/catkin_workspace/install_isolated/env.sh', 'make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /tmp/catkin_workspace/build_isolated/moveit_ros_planning && /tmp/catkin_workspace/install_isolated/env.sh make -j1

Command failed, exiting.
# END SUBSECTION
$ 
```

dornhege, 2016-04-13:
This was added in moveit_msgs, which would need to be included in a new release.

130s, 2016-04-16:
@dornhege thanks for the tip. Dependency between packages (or should I say between repositories) seemed not clear to me.
Now done https://github.com/ros/rosdistro/pull/11118

closed 2016-04-16 by 130s