moveit/moveit_ros #676: The catkin_package LIBRARIES list is missing moveit_collision_plugin_loader

Levi-Armstrong, 2016-05-23:
This is true for both indigo-devel and jade-devel branches.

[Library](https://github.com/ros-planning/moveit_ros/blob/jade-devel/planning/collision_plugin_loader/CMakeLists.txt#L1)
[Missing here](https://github.com/ros-planning/moveit_ros/blob/jade-devel/planning/CMakeLists.txt#L47-L55)

rhaschke, 2016-05-24:
Actually, the library is implicitly linked into [moveit_planning_scene_monitor](https://github.com/ros-planning/moveit_ros/blob/jade-devel/planning/planning_scene_monitor/CMakeLists.txt#L9), which in turn [is listed in catkin_package](https://github.com/ros-planning/moveit_ros/blob/jade-devel/planning/CMakeLists.txt#L55).
What is the actual error you get?

Levi-Armstrong, 2016-05-24:
I am getting the error: undefined reference to `collision_detection::CollisionPluginLoader::CollisionPluginLoader()'. I downloaded the source and made the change and it compiled without any errors. I did not try running it with the source without the change, but I tried it this morning and it also compiled without errors.

So my code compiling using the source but not with the binary.

rhaschke, 2016-05-24:
Could you please provide more context information:
What are you compiling? Whole MoveIt workspace? Parts of it? Your own project?
Which build environment do you use? OS? Compiler? Linker?
As you can see on [Travis CI](https://travis-ci.org/ros-planning/moveit_ros), `moveit_ros` compiles without issues from source.

Levi-Armstrong, 2016-05-24:
>What are you compiling?

I am compiling a custom package which is using CollisionPluginLoader. Because of the compiler errors I then cloned the moveit_ros repository and compiled the whole workspace which includes my custom package and moveit_ros and the compiler errors went away.

>Which build environment do you use? OS? Compiler? Linker?

I am using catkin_make on Ubuntu 14.04.
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 

>As you can see on Travis CI, moveit_ros compiles without issues from source.

The issue is not with the source, but possible with the currently released binary. It is possible that it is something with my package but if it was I would have expected it not to compile using moveit_ros source also.

Levi-Armstrong, 2016-05-24:
@rhaschke ,

My current CMakeLists.txt file is below. I have added the moveit_ros_planning to my find_package and after words I print out all of the libraries and moveit_collision_plugin_loader.so is not listed. Now this is using the binary, but this works if I include moveit_ros package withing my workspace.  

I was able to get it to work with the binary but I had to add the line below just after the find_package(catkin ...). Do you have any thoughts on why when adding moveit_ros_planning to the find_package it is not adding the libmoveit_collision_plugin_loader.so to the catkin_LIBRARIES?

```
list(APPEND catkin_LIBRARIES /opt/ros/indigo/lib/libmoveit_collision_plugin_loader.so)
```

Current CMakeList.txt:
```
cmake_minimum_required(VERSION 2.8.3)
project(industrial_moveit_benchmarking)

add_definitions("-std=c++11")

## Find catkin macros and libraries
## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
## is used, also find other catkin packages
find_package(catkin REQUIRED COMPONENTS
  roscpp
  moveit_ros_planning
  stomp_moveit
  constrained_ik
  cmake_modules
  pluginlib
  moveit_core
)

MESSAGE( WARNING ${catkin_LIBRARIES})

## System dependencies are found with CMake's conventions
find_package(cmake_modules REQUIRED)
find_package(Boost REQUIRED)
find_package(Eigen REQUIRED)

catkin_package()

###########
## Build ##
###########

include_directories(include ${catkin_INCLUDE_DIRS} ${Boost_INCLUDE_DIRS} ${EIGEN_INCLUDE_DIRS})

## Declare a C++ executable
add_executable(stomp_benchmarking_node src/stomp_valgrind.cpp)
add_executable(joint_interpolated_benchmarking_node src/joint_interpolated_valgrind.cpp)

## Specify libraries to link a library or executable target against
target_link_libraries(stomp_benchmarking_node ${catkin_LIBRARIES})
target_link_libraries(joint_interpolated_benchmarking_node ${catkin_LIBRARIES})
```

To solve the issue using the binary I had to explici

Levi-Armstrong, 2016-05-24:
>Actually, the library is implicitly linked into moveit_planning_scene_monitor, which in turn is listed in catkin_package.

I think the above state would only work if moveit_planning_scene_monitor was a catkin package and it exposed the library to catkin using catkin_package(...). Since it is not a catkin package I believe it still needs to be listed in the planning packages CMakeLists.txt to expose the library to other catkin packages. Is this correct?

Levi-Armstrong, 2016-05-24:
I have done further testing and when including the moveit_ros source in my workspace and building using catkin_make it will build fine because the library is in my local workspace but if I perform a catkin_make_isolate if fails to build with the same errors. Now if I add the library as described in the original comment it compiles using catkin_make_isolate. I am certain now that this library need to be included so the library is exported to other catkin packages.

rhaschke, 2016-05-24:
Thanks for your intense testing. Indeed, using catkin_make all sources in the workspace are build within one cmake universe. Hence, all targets are known and the library is found.
Building against binary package or using catkin_make_isolated separates the cmake environments for each package and the target is not known anymore. I will merge your PR. Thanks again for your efforts.

closed 2016-05-24 by rhaschke

Levi-Armstrong, 2016-05-25:
>Thanks for your intense testing.

@rhaschke My pleasure. I learned something new, I was not aware of catkin_make_isolate.

Also, I looked at the other libraries created in the moveit_ros_planning package and there are three other libraries that are created that are not listed, which are shown below. Should these also be added? In the past I have found it useful to use moveit plugins outside of moveit for compatibility. If so let me know and I will create a new issue/pull request.

```
moveit_kdl_kinematics_plugin
moveit_default_planning_request_adapter_plugins
moveit_srv_kinematics_plugin
```

rhaschke, 2016-05-25:
plugin libs shoudn't be listed in catkin_package(). They must not be linked at compile time, but only loaded dynamically at runtime.