moveit/moveit_ros #681: Jade failures

wjwwood, 2016-06-06:
There are several regressions related to moveit on the ROS Jade build farm at the moment. They seem to have been failing for some time, and right now they're blocking our ability to sync the Jade changes to public. Here's the list of regressions (that is to say packages which are currently publically available, but are currently failing on the farm):

http://repositories.ros.org/status_page/ros_jade_default.html?q=REGRESSION

There are several tickets across this repo and the `moveit_core` repo which seem to be about planning of releases, but I wasn't sure which to post this on. Feel free to close this and point me to another issue.

It would help us a lot if you guys could release a version that builds (even if it is a short term fix), that way we can sync other fixes to public without removing existing moveit debs in the process.

Thanks!

mikeferguson, 2016-06-07:
Looks like the blocker is moveit_ros_planning http://build.ros.org/job/Jbin_uT64__moveit_ros_planning__ubuntu_trusty_amd64__binary/ -- build failure appears to be related to the shape_tools changes -- @wjwwood -- I'll try to look into this later today

mikeferguson, 2016-06-07:
Appears this was fixed in https://github.com/ros-planning/moveit_ros/pull/587 -- but has not yet been released -- not sure how the shape_tools dependency got lost (as the file is still there, but package has not been installed) so it must have been transitive from a lower package. I'm trying to resolve the issues with https://github.com/ros-planning/moveit_ros/pull/679 and then I'll push a new release

wjwwood, 2016-06-07:
Thanks @mikeferguson!

mikeferguson, 2016-06-09:
while I know https://github.com/ros/rosdistro/pull/11780 won't build all packages on all platforms, this should get us back to no regressions

mikeferguson, 2016-06-09:
Looks like at least the first deb has succeeded again: http://build.ros.org/job/Jbin_uT64__moveit_ros_planning__ubuntu_trusty_amd64__binary/35/

closed 2016-06-09 by mikeferguson