moveit/moveit_ros #682: `move_group` dies when service `/get_planner_params` is called

nikhilkalige, 2016-06-07:


rhaschke, 2016-06-07:
Can you provide sample code that fails? On my side, retrieving and displaying the planner params in rviz works like a charm.

nikhilkalige, 2016-06-07:
I wrote some 50 lines in the issue above, looks like it disappeared somewhere... 

I tried the following code with 
```
roslaunch ur_gazebo ur5.launch
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
```
I then queried with the following 
``` 
rosservice call /get_planner_params RRTkConfigDefault manipulator
```
I also tried the python interface
```
import rospy
from moveit_msgs.srv import GetPlannerParams, GetPlannerParamsRequest
rospy.wait_for_service('/get_planner_params')
s = rospy.ServiceProxy('/get_planner_params', GetPlannerParams)
h = GetPlannerParamsRequest()
h.group = 'manipulator'
h.planner_config = 'BKPIECEkConfigDefault'
s(h)
```

rhaschke, 2016-06-07:
Good catch. The validity check is indeed missing. Created PR #684 to fix it.

nikhilkalige, 2016-06-07:
Awesome.. Works properly now..

commit eda6a389 (2016-06-07): Merge pull request #684 from ros-planning/fix#682

added missing validity check (fixes #682)
closed 2016-06-07 by nikhilkalige