moveit/moveit_ros #689: indigo/jade releases

v4hn, 2016-06-16:
given that we have accumulated a number of bug fixes and the new ApplyPlanningScene service,
it would be great to have a new release of moveit_core and moveit_ros.

Are there any particular requests/issues we should wait for?

130s, 2016-06-16:
Something that may be nice to discuss; One thing we could do for the long existing PRs is to manually rebase them and at least have them go through CI tests.
Even we decided to do so, I'm still not sure if now is the right timing for doing so since it'll require resource (reviewers, time).

davetcoleman, 2016-06-16:
+1 to releasing

The long existing PRs seem unrelated to the decision to release or not, no?

130s, 2016-06-17:
Well, one way to think is to process as many PRs as possible before the release.

But I guess we can also make smaller releases with more frequency, if that's what we're aiming. Then +1 for releasing.

v4hn, 2016-06-17:
On Thu, Jun 16, 2016 at 10:11:24PM -0700, Isaac I.Y. Saito wrote:
> But I guess we can also make smaller releases with more frequency,
> if that's what we're aiming.

I suppose we should, if we want to have "regular" users use the deps
instead of their own source builds.

+1 that makes three of us.

davetcoleman, 2016-06-17:
I never thought of it that way!

130s, 2016-06-21:
- Indigo 0.7.2 is out https://github.com/ros/rosdistro/pull/11899.
- For Jade (and presumably Kinetic), at least https://github.com/ros-planning/moveit_ros/pull/607 needs resolved afaik.

davetcoleman, 2016-08-04:
@130s Can we consider Jade "released" yet? Should we add installation instructions to the website?

Can we close this issue? Kinetic discussions are in other issues.

davetcoleman, 2016-08-06:
jade discussion moved to https://github.com/ros-planning/moveit/issues/22

closed 2016-08-06 by davetcoleman