moveit/moveit_ros #692: Release *-moveit-full metapackage

davetcoleman, 2016-06-22:
I was working on the MoveIt! Docker builds just now and noticed there is no ``ros-jade-moveit-full`` released. Am I missing something or do you know how to release those metapackages? It exists for indigo already. @IanTheEngineer @mikeferguson @v4hn 

Thanks!

mikeferguson, 2016-06-22:
I think the metapackages aren't released since the things below them aren't building on most platforms (when we are ready, the repo is over here https://github.com/ros-planning/moveit_metapackages)

It looks like our warehouse_ros debbuilds are still failing on >trusty, but I haven't gotten a chance to look into why

130s, 2016-06-22:
> our warehouse_ros debbuilds are still failing on >trusty, but I haven't gotten a chance to look into why

Do you have links to them? I don't see build issues with I and J.
http://repositories.ros.org/status_page/ros_indigo_default.html?q=warehouse
http://repositories.ros.org/status_page/ros_jade_default.html?q=wareh

mikeferguson, 2016-06-22:
Huh -- I had gotten a bunch of emails about build failures -- but you are right, they appear to now be building -- hurray!

davetcoleman, 2016-06-22:
Great - so we can release moveit-full?

davetcoleman, 2016-06-23:
FYI also missing from Jade: moveit_setup_assistant and moveit_ikfast

mikeferguson, 2016-06-23:
Ok -- so the buildfailures I'm seeing are from "moveit_ros_warehouse" not warehouse_ros, and it is because we haven't yet merged and released some version of https://github.com/ros-planning/moveit_ros/pull/607

130s, 2016-06-26:
> Ok -- so the buildfailures I'm seeing are from "moveit_ros_warehouse" not warehouse_ros, and it is because we haven't yet merged and released some version of #607

I now see. I also see that that's [about Jade](http://repositories.ros.org/status_page/ros_jade_default.html?q=warehouse) (for Indigo it seems fine).

davetcoleman, 2016-08-04:
@130s can you move this discussion to the new repo under the title "Jade Release" and close this issue?

davetcoleman, 2016-08-09:
The plan is to not release ``*-moveit-full`` metapackage for jade or kinetic, instead replacing with simply ``*-moveit`` that reflects the new repo. Closing

closed 2016-08-11 by davetcoleman