moveit/moveit_ros #695: Kinetic release

davetcoleman, 2016-06-24:
I'm working on getting the [experimental merged repo for Kinetic](https://github.com/davetcoleman/moveit) to compile and I'm noticing a lot of missing dependencies. I'm opening this issue to provide a place to track those dependency issues.

davetcoleman, 2016-06-24:
FCL blocker: https://github.com/ros-planning/moveit_core/issues/299

davetcoleman, 2016-06-27:
@tfoote how do we go about getting "abandoned" ROS message repos released? Is anyone maintaining https://github.com/ros-interactive-manipulation/ ?

We need the follow two message package released:
https://github.com/ros-interactive-manipulation/
https://github.com/ros-interactive-manipulation/household_objects_database_msgs

tfoote, 2016-06-27:
@davetcoleman Please open a ticket on the repos you need released. 

@jkammerl Could you add me to the ros-interactive-manipulation organization. You're the only publicly listed member. [Our policy](http://wiki.ros.org/RecommendedRepositoryUsage/CommonGitHubOrganizations#Guidelines_for_administering_ros-.2A_organizations_on_GitHub) is that the ROS platform manager is an owner on all `ros-*` repos to support maintainer handoffs.

davetcoleman, 2016-06-27:
I have opened tickets there, see first two issues referenced inside this issue

davetcoleman, 2016-06-27:
Also I believe @isucan is a member of ros-interactive-manipulation

davetcoleman, 2016-06-27:
Required to build from source: https://github.com/ros-planning/moveit_ros/pull/698

davetcoleman, 2016-07-05:
Moved ownership of ros-interactive-manipulation conversation to https://github.com/ros-interactive-manipulation/tabletop_object_perception/issues/3

davetcoleman, 2016-07-08:
MoveIt! Kinetic is now passing on Travis, though requires these two PRs: https://github.com/ros-planning/moveit_core/pull/304, https://github.com/ros-planning/moveit_ros/pull/699.

Travis also passes for unified MoveIt! repo: https://github.com/davetcoleman/moveit

dbolkensteyn, 2016-07-26:
Can anyone else reproduce this issue? If so it'd be a blocker IMO. https://github.com/ros-planning/moveit_setup_assistant/issues/119

davetcoleman, 2016-08-04:
Moving this issue to new repo: https://github.com/ros-planning/moveit/issues/18

closed 2016-08-04 by davetcoleman