moveit/moveit_ros #712: mesh_filter test fails and hangs. Test is disabled on CI builds because there is no display available.

scottpaulin, 2016-07-14:
The mesh_filter test continually fails on `EXPECT_FLOAT_EQ (filtered_depth [idx], gt_depth [idx]);` and doesn't seem to terminate when run locally when there is a display available. I had a look at the output of one of @davetcoleman 's CI builds and it seems that this test is disabled in CI builds because there is no display available (see line 420 "No display, will not configure tests for moveit_ros_perception/mesh_filter" of the CI output at [1]). 

The tests are disabled with line 18 of moveit_ros/perception/mesh_filter/CMakeLists.txt.

OS: Ubuntu 16.04.
ROS: Kinetic binaries with source install of Moveit. 

[1] https://travis-ci.org/ros-planning/moveit_core/builds/143428552

davetcoleman, 2016-07-14:
Seems like there are two issues here:

1) there is a bug in mesh_filter or there is a bad test
2) CI does not have a display available in Docker. This could be fixed in [moveit_ci](https://github.com/ros-planning/moveit_ci) using [these instructions](http://wiki.ros.org/docker/Tutorials/GUI). 

Either pull request would be welcomed!

scottpaulin, 2016-07-23:
There is now a [pull request](https://github.com/ros-planning/moveit_ci/pull/1) in moveit_ci to allow displays to be used in the ci. The xvfb changes shown in the readme would need to be implemented in this repo's travis.yml file. 

I have made these changes in my fork of moveit_ros. There seems to be a build error in the mesh_filter tests (they previously weren't being built) which I will fix.