moveit/moveit_ros #736: No planning scene updates in Rviz after #716.

dornhege, 2016-08-03:
I run a robot in simulation, start move_group and Rviz. The planning scene initially appears, but when moving the robot head via teleop is never updated. The robot state doesn't change and the Octomap isn't updated. This seems quite basic, can someone confirm this is an issue?

If I go back to before #716 everything works as expected. I haven't followed, what exactly is happening there, but as it deals with updates I suspect, but cannot confirm that this is the cause.
The newPlanningSceneCallback seems to never be called, although rostopic hz shows messages on the monitored_planning_scene topic.

I've tried with indigo-devel and also merging in #728 in its current state.

v4hn, 2016-08-03:
do you have a simple way by which I could reproduce this bug?

dornhege, 2016-08-04:
The most objective thing I have found besides "looking" is to put a printout in planning_scene_monitor.cpp newPlanningSceneCallback. Then just run moveit and the RViz plugin. This printout should (as far as I understand everything) always come, when things are working correctly. To be safe put `rostopic hz` on the topic set in rviz (i.e. /move_group/monitored_planning_scene).
Compare indigo-devel and ros-planing/moveit_ros@ae51d07d5b1b848d87acb8c7dacb078b17d9c5ca.

dornhege, 2016-08-04:
@v4hn : The comment in #728 about the spinner might be the reason for this. The newPlanningSceneCallback is called via the spinner. If that is in the "wrong" thread and something doesn't work as expects, message might never be dispatched.

dornhege, 2016-08-04:
Digging through RViz warnings I found this:

[1470311817.138041964, 2197.807000000] ros.roscpp: AsyncSpinnerImpl: Attempt to start() an AsyncSpinner failed because another AsyncSpinner is already running. Note that the other AsyncSpinner might not be using the same callback queue as this AsyncSpinner, in which case no callbacks in your callback queue will be serviced.

v4hn, 2016-08-04:
On Thu, Aug 04, 2016 at 05:10:29AM -0700, Christian Dornhege wrote:
> Digging through RViz warnings I found this:
>
> [1470311817.138041964, 2197.807000000] ros.roscpp: AsyncSpinnerImpl: Attempt to start() an AsyncSpinner failed because another AsyncSpinner is already [...]

I suppose this means seals the revert question then.
I'll have a look at it this evening.

dornhege, 2016-08-05:
Fixed with #742 .

closed 2016-08-05 by dornhege