moveit/moveit_ros #745: How do I set MoveIt! to use the source installation  ompl for ros indigo?

gxxzs, 2017-09-20:
I install moveit from source for ros indigo according to the steps did in ros-planning/moveit_ros#623, How do I set MoveIt! to use the source installation of ompl for ros indigo? After i remove ros-indigo-ompl and install ompl from source, moveit cannot load ompl unless I do "sudo apt-get install ros-indigo-ompl" or “”sudo apt-get install ros-rosversion -d-ompl”.
My installayion process:

    1.Installed MoveIt from source and "sudo apt-get remove ros-indigo-ompl" to remove ompl after moveit installed successfully.
    2.MoveIt path in my ubuntu was:~/ws_moveit
    Installation Link:http://moveit.ros.org/install/
    Installation link:http://picknik.io/moveit_wiki/index.php?title=OMPL/Add_New_Planner
    http://moveit.ros.org/install/source/dependencies/
    3. install ompl from source
    cd ~/catkin_ws/src
    git clone https://github.com/ompl/ompl.git
    cd ompl/
     wget https://raw.githubusercontent.com/ros-gbp/ompl-release/debian/indigo/trusty/ompl/package.xml
   cd ~/catkin_ws
    catkin_make_isolated

Now moveit cannot load ompl, 
when i run "roslaunch pi_robot_moveit_config demo.launch" 
[ERROR] [1505903843.188977089]: Exception while loading planner 'ompl_interface/OMPLPlanner': Failed to load library /home/zs/ws_moveit/devel/lib/libmoveit_ompl_planner_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libompl.so.10: cannot open shared object file: No such file or directory)
Available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner

i konw something should be do to link ompl to moveit, but I donot know how, can anyone give me some advice, thanks.

davetcoleman, 2017-09-21:
Please see source installation instructions for OMPL [here](http://moveit.ros.org/install/source/dependencies/). Note that they are for ROS Kinetic.

gxxzs, 2017-09-25:
http://moveit.ros.org/install/source/dependencies/
I have installed  OMPL from source and  correct the "wget https://raw.githubusercontent.com/ros-gbp/ompl-release/debian/indigo/trusty/ompl/package.xml" in ros indigo ,but the moveit still cannot load ompl. Did i have to turn to the  Ros Kinetic?

gxxzs, 2017-09-30:
I run the  catkin build as http://moveit.ros.org/install/source/dependencies/ something still wrong with it
zs@zs:~/ompl$ catkin build
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/zs/catkin_ws/devel:/home/zs/ws_moveit/devel:/opt/ros/kinetic
Workspace:                   /home/zs/ompl
--------------------------------------------------------------------------------
Source Space:       [exists] /home/zs/ompl/src
Log Space:         [missing] /home/zs/ompl/logs
Build Space:        [exists] /home/zs/ompl/build
Devel Space:        [exists] /home/zs/ompl/devel
Install Space:      [unused] /home/zs/ompl/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] No packages were found in the source space '/home/zs/ompl/src'
[build] No packages to be built.
[build] Package table is up to date.                                           
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 1.3 seconds ]             
[build] Summary: All 1 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 1.3 seconds total.                                            
zs@zs:~/ompl$ catkin build
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/zs/catkin_ws/devel:/home/zs/ws_moveit/devel:/opt/ros/kinetic
Workspace:                   /home/zs/ompl
--------------------------------------------------------------------------------
Source Space:       [exists] /home/zs/ompl/src
Log Space:          [exists] /home/zs/ompl/logs
Build Space:        [exists] /home/zs/ompl/build
Devel Space:        [exists] /home/zs/ompl/devel
Install Space:      [unused] /home/zs/ompl/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] No packages were found in the source space '/home/zs/ompl/src'
[build] No packages to be built.
[build] Package table is up to date.                                           
[build] Summary: All 0 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 0.0 seconds total.